sparkfun/Arduino_Boards #5: The ESP8266 branch is not loading the compiler correctly.

JamesHagerman, 2015-06-07:
First, I tried adding the SparkFun Board Manager json file mentioned on the ESP8266 tutorial and it didn't work correctly. (this is the broken link: https://raw.githubusercontent.com/sparkfun/Arduino_Boards/esp8266/IDE_Board_Manager/package_sparkfun_index.json)

Because that didn't work, I downloaded this repo instead, switched to the ESP8266 branch, and copied the hardware directory to my Sketchbook's hardware directory. After that, I selected the ESP8266 Thing from the board menu, set up my FTDI port, copy and pasted the blink demo code, and hit compile.

It gave me this error:
Cannot run program "{runtime.tools.xtensa-lx106-elf-gcc.path}/bin/xtensa-lx106-elf-g++" (in directory "."): error=2, No such file or directory

Still, I was able to get it to compile by adding the "official" ESP8266 for Arduino JSON file from this repo in the Arduino 1.6.4 preferences panel:
https://github.com/esp8266/Arduino

Thanks!

JamesHagerman, 2015-06-07:
Oops! I just realized that I had a space in the Preferences panel's Board Manager text box. Apparently spaces mess the thing up! So, the tutorials json DOES work!

Still, this repo still seems to have an issue.

BrentWilkins, 2015-06-16:
It works for you, so it's not clear w/o more information what the issue is.  Do you have more details, so we can address the issue?

ToniCorinne, 2016-01-08:
Closing this issue out. The branch has been merged into master. If there are additional difficulties, please open up a new issue.

closed 2016-01-08 by ToniCorinne