sparkfun/Arduino_Boards #10: Board [***] (platform avr, package SparkFun) is unknown

benkuper, 2015-11-05:
Hi,
i just updated both Arduino to 1.6.6 and the sparkfun boards definition to 1.0.3 (as proposed by the Arduino IDE), and now i can't compile for any of the sparkfun boards.

For instance if i choose makey makey, i get this error : 
> Board makeymakey (platform avr, package SparkFun) is unknown

Any idea on what's going down ?

Thanks

BrentWilkins, 2015-11-05:
Looks like a fun bug... I was able to see an equivalent error for another board in a separate repository. While poking around trying to track down that issue it went away, and I cannot reproduce it. I then tried the MaKey MaKey, but never got the error.

Will you try building Blink for the MaKey MaKey and share any errors? What OS are you using? There seem to be known, open issues that vary from platform to platform in this version of the IDE.

benkuper, 2015-11-05:
Same for me actually, i just removed it, checked that all sparkfun related folder was removed from the packages folder and reinstalled it, it's now working.
It may be a bug just for people updating Arduino when already having the sparkfun definitions.

Anyway, thanks !

martinmm, 2015-11-15:
The instructions given in this github repository work for me with Linux, the addon zip file from the hookup instructions do not

https://learn.sparkfun.com/tutorials/pro-micro--fio-v3-hookup-guide

me@home:~/arduino-1.6.6$ ./arduino
Picked up JAVA_TOOL_OPTIONS: 
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

Sketch uses 3,912 bytes (13%) of program storage space. Maximum is 28,672 bytes.
Global variables use 148 bytes of dynamic memory.
java.io.IOException: Cannot run program "{path}/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:113)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:157)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1178)
	at processing.app.Sketch.exportApplet(Sketch.java:1152)
	at processing.app.Sketch.exportApplet(Sketch.java:1124)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2452)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.cmd.path={runtime.ide.path}/hardware/tools/avr/bin/avrdude, automatically converted to tools.avrdude.cmd.path={path}/bin/avrdude. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated tools.avrdude.config.path={runtime.ide.path}/hardware/tools/avr/etc/avrdude.conf, automatically converted to tools.avrdude.config.path={path}/etc/avrdude.conf. Consider upgrading this core.
Warning: platform.txt from core 'SparkFun AVR Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

ToniCorinne, 2016-01-08:
The hookup guide has been updated, so I'm closing this issue out.

closed 2016-01-08 by ToniCorinne