sparkfun/Arduino_Boards #12: Arduino 1.6.7: Board promicro (platform avr, package SparkFun) is unknown  Error compiling.

BartmanEH, 2016-01-22:
I have uninstall Arduino IDE 1.6.7, re-installed it, unistalled the SparkFun board package, re-installed it but I still get this compile error.

I noticed that board manager library files don't seem to go into the Arduino IDE program files installation folder but I'm not sure if this was an IDE change, bug or related to this issue.

BrentWilkins, 2016-01-22:
We experience a variety of issues with versions of the IDE greater than 1.6.5 & there are references to this in the official Arduino GitHub repository and various forums. Until Arduino fixes this we recommend using IDE version 1.6.5-r5 because it works. It's a tradeoff because it has other bugs, but I get work done in it.

closed 2016-01-22 by BrentWilkins

BartmanEH, 2016-01-22:
Well this is a serious impediment for me. I spent a lot of effort to get various projects to compile with IDE v1.6.6. To overcome this problem, I uninstalled it and installed 1.6.5 and indeed can get the SparkFun board support package to work but now my projects don't compile. What a pain.

BrentWilkins, 2016-01-22:
I'd be curious to know what will only compile in versions newer than 1.6.5. 1.6.5 through 1.6.7 all use the same version of gcc.

One of the errors I've experienced with the broken version(s) of Arduino is that it will download the correct json file, but for whatever reason not download and install the archive referenced in it. If you want to try and manually install our board files you could install [this directory](https://github.com/sparkfun/Arduino_Boards/tree/master/sparkfun) on your system.

You can also manually do what the board manager does. Here is what Arduino defines what I'll refer to as the base directory:
* **GNU/Linuxes:** ~/.arduino15/
* **Windows:** %APPDATA%\Arduino15
* **Mac OSX:** ~/Library/Arduino15/

Inside this base directory you should find all of the following files:
* <base>/package_sparkfun_index.json
* <base>/package_sparkfun_index.json.sig.tmp
* <base>/staging/packages/sparkfunboards.1.1.2.tar.bz2
* <base>/pacakges/SparkFun

Work your way down the list. If the json isn't there, you can find it [here](https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json). It currently links to [this bzip tarball](https://github.com/sparkfun/Arduino_Boards/raw/master/IDE_Board_Manager/sparkfunboards.1.1.2.tar.bz2). You can download that into staging/packages, but might not need to.
You can extract that into the packages directory.

BartmanEH, 2016-01-22:
Thanks, Brent. I'll work through the manual board package with v1.6.7 as you suggest.

There seems to be problematic library handling in v1.6.5 - I get compile errors about missing .h files that are in a library subfolder but don't seem to be seen by v1.6.5 IDE but are found just fine by v1.6.6/7. Specifically LiquidCrystal_I2C.h - I've deleted conflicting libraries in the program's "C:\Program Files (x86)\Arduino\libraries" library folder but the problem remains. Strangely I can solve missing file compile errors one at time by copying in appropriate folders into the library folder until this one.

BartmanEH, 2016-01-22:
All your referenced files are there - ignoring the typo for "/pacakges/SparkFun" and asssuming "/packages/SparkFun" :-) and the folder %APPDATA%\Arduino15\packages\SparkFun\hardware\avr\1.1.2 exists but I still get:
Arduino: 1.6.7 (Windows 10), Board: "SparkFun Pro Micro, ATmega32U4 (5V, 16 MHz)"
Board promicro (platform avr, package SparkFun) is unknown
Error compiling.

BartmanEH, 2016-01-24:
I found a work around <a href="https://github.com/arduino/Arduino/issues/4137#issuecomment-160010782" target="_blank">here</a>: by selecting another SparkFun board until I found one that would compile (Makey Makey in my case) and then going back to Pro Micro, I was able to compile without a board missing error. There's something flaky with Arduino IDE 1.6.6/7 and hardware board library management since this problem doesn't occur with 1.6.5r5.

slootsky, 2016-01-24:
For the missing    . h  files in the 1.6.5 ide, I had the same issue yesterday, and I found that if I included the   h   in my main   .ino, then when it was included in another   .h, it was found.

BartmanEH, 2016-01-24:
@slootsky thanks for that tip and I'll keep it in mind for future problems, however, I really loath going way back to 1.6.5 - with the work around I've found, I'm able to move forward with 1.6.7 and even nightly builds and have fewer problems overall.

freespace, 2016-02-28:
I experienced this error on OS X, and purging ~/Library/Arduino* and reinstalling the SparkFun boards fixed the problem. This indicates an issue with compatibility between boards installed by different versions of Arduino.

jamieisboss, 2016-06-11:
All of these issues are closed, but the problem isn't resolved without manual intervention. 

Might it be worth putting a note in the wiki with instructions as to how to resolve this issue, and probably link that from the readme for a little while, not everyone knows to look at closed github issues for a fix and until the arduino ide tells the user what to do, or removes and reinstalls the affected files, i wouldn't consider this to be resolved. 

IM(not all that)HO, issues like this are easy to consider resolved one the technical fix is found, but that doesn't mean anything to 85% of users, and it's the users not being able to do the thing that is the problem, not the underlying technical issue. While the arduino IDE probably won't fix this in coming versions, as I've just experienced the issue on 1.6.9 and so it seems to have been an issue for a while, i cant see any clear instructions a novice could follow to resolve the problem they're having, either in a readable and obvious note here on github, or anywhere obvious on the sparkfun website.

Years of support instinct says I should look for the perspective of a user who got this working in 1.6.4 for the first time ever, then it just stopped working. I wouldn't accept this issue was resolved from a team i was managing until that user knows there's an issue and what they can do to fix it.

Looking back, the package has been broken for over a year, why are you still telling people to download it, there's no bug open in the arduino IDE issues list, surely this isn't the level of support and service sparkfun want to be providing?

drelephant, 2018-03-10:
I updated from 1.6.5 to 1.8.5 but then no matter what I did, I got the error:
> Board promicro (platform avr, package SparkFun) is unknown
> Error compiling.

I found that I had folders in both
`C:\Users\myusername\AppData\Local\Arduino15`
and also
`C:\Users\myusername\AppData\Roaming\Arduino15`

What fixed it for me was deleting both those Arduino15 folders, uninstalling the Arduino software and reinstalling.