sparkfun/Arduino_Boards #13: Sparkfun Pro Micro Hookup Guide out of date

MortimerNeuss, 2016-02-02:
The Sparkfun Pro Micro Hookup Guide links to this repo with the words: " It should be a folder named ‘SFE32U4_Drivers’.". Apparently this repo has changed since and I guess there is not a Windows x64 Driver anymore to support the Sparkfun Pro Micro v5?

ToniCorinne, 2016-02-02:
No, they are still in this repo. They are just buried in [here](https://github.com/sparkfun/Arduino_Boards/tree/master/sparkfun/avr/bootloaders/caterina). With the new Arduino board manager, it was easier to have all of our boards in one big batch file to install, instead of several different repositories. Sorry about the confusion! We'll get the tutorial updated!

ToniCorinne, 2016-02-03:
Tutorial has been updated - closing this out.

closed 2016-02-03 by ToniCorinne

garyhill, 2016-04-20:
Same question as original post. I have looked in the "The Sparkfun Pro Micro Hookup Guide" - "Installing: Windows" and still see the "It should be a folder named ‘SFE32U4_Drivers" Have searched entire Sparkfun repository and can not find this folder or any driver for the 32U4.

BrentWilkins, 2016-04-21:
I've updated the guide a little. The correct path in the zip archive linked to is Arduino_Boards-master → sparkfun → avr → signed_driver.

garyhill, 2016-04-21:
By downloading the "signed_driver" folder and then pointing Windows to this folder the device driver was discovered and installed. Confusion on my part was that there is no folder named SFE32U4_Drivers in this folder and in fact I could find no drivers in this folder - even though Windows clearly did. It might help others to tell them which file Windows is looking for (dpinst-amd64, dpinst-x86, qtechknow, Qtechknow, sparkfun,...) in this folder.

BrentWilkins, 2016-04-21:
I don't have a Windows machine handy right now, so I only pointed them to sparkfun.inf in the guide. I know that can be used, but I'm not able to test the exact instructions in the guide tonight.

Rom3oDelta7, 2016-04-21:
Brent,

I had no problem finding it - the problem is that it will not install on Win8, 8.1, or 10 (at least for x64). You get the message:
**The hash file for the file is not present in the specified catalog file. This is likely corrupt or the victim of tampering.**

I've tried selecting the sparkfun.inf file directly in Update Driver, tried running dpinst-amd64.exe in various compatibility modes, and many other futile attempts.

I have posted on the forum, sent email to support, but no response. Since upgrading from win7 I have been unable to use my Pro Micros. Would really appreciate it if Sparkfun could fix this. Thanks.

Rob

BrentWilkins, 2016-04-21:
Have you tried right-clicking on the INF, and selecting 'Install'? I believe that's how I would do it, but I can't reasonably verify that right now.

Rom3oDelta7, 2016-04-21:
Yes. You get the same error message. 
I forgot that you get a different error in win10/x64. The ProMicro installs with a generic "USB Serial Device" driver and when you try to update the driver it says the best driver is already installed. Running dspinst-amd64.exe on Win10/x64: the Qdino driver installs OK but the "Sparkfun Electronics (usbuser)" driver just says "Install failed" just like on Win8.

Rom3oDelta7, 2016-04-21:
I just tried something else. 
1. I selected Sparkfun Pro Micro as the board in the IDE
2. I forced the driver to be an Arduino Micro in Update Driver/Let me pick ...
3. The IDE uploaded the sketch (blink) and it seems to work. In the process, the device was remapped to an Arduino Leonard Bootloader and then back to an Arduino Micro (or sometimes back to the generic USB Serial Device") which I assume is the proper behavior for the 32U4 bootloader they share.

It's a bit of a kludgy workaround, but at least I can load software onto it now. Sure would be nice to avoid this and install a working driver though. Thanks for your help.

RD7

BrentWilkins, 2016-04-21:
I sure hope you did that on a 5V Pro Micro. Uploading code for a 16MHz 32U4 can cause real issues on  an 8MHz 32U4.

Rom3oDelta7, 2016-04-21:
Indeed! I have both, but this one is 5V. I put labels on them to make sure I don't make that mistake.
BTW, I just tested it and this workaround works on Win8 as well, although it is a bit more work to force the driver to be an Arduino Micro. I assume by your comment that the bootloader is processor-aware and that this should not be attempted AT ALL on a 3.3V ProMicro? Thanks - really happy to finally be having a conversation about this!

RD7

garyhill, 2016-04-21:
Did not know my question would generate so much interest. For the record I am working with a 3.3v ProMicro so the work around does not sound like it will work for me. On the plus side one of my computers is running Windows 7. The bad news is the other one is Windows 10.

Rom3oDelta7, 2016-04-21:
I'm glad it did! I have been stuck for a long time on this. I've been monitoring my various forum posts and this is the first one that got a reply from Sparkfun (thanks again @BrentWilkins!)

We still need SparkFun to fix the signed driver and make all this go away. I suspect that this is not a difficult fix, but while I am a developer I am not fluent in Windows device drivers, so what do I know? ;-)

RD7

Rom3oDelta7, 2016-04-21:
ALL - I just noticed a post @markeon made on the Sparkfun forum with a much better workaround involving turning off Windows signed driver validation. I tried it on Win10/x64 and it works (but be patient with the process, and it involves a lot of reboots). The forum topic is [here ](https://forum.sparkfun.com/viewtopic.php?f=32&t=42942) and the Sparkfun tutorial on disabling the signature check is [here](https://learn.sparkfun.com/tutorials/di ... -windows-8).

So it looks like we finally have a suitable workaround for all of the variables involved in this! Thanks again to **@markeon** for this post.

RD7

Rom3oDelta7, 2016-04-22:
Heard back form Sparkfun tech support today and they sent me a working x64 ProMicro driver. See the [most recent post on this forum thread](https://forum.sparkfun.com/viewtopic.php?f=32&t=42942).

Enjoy!

RD7