sparkfun/Arduino_Boards #15: sparkfunboards v1.1.4 borken

stejsoftware, 2016-04-23:
I updated sparkfunboards from v1.1.3 to v1.1.4 and now get this.

```Arduino:` 1.6.8 (Windows 7), Board: "SparkFun Pro Micro, ATmega32U4 (3.3V, 8 MHz)"

Board promicro (platform avr, package SparkFun) is unknown

Error compiling for board SparkFun Pro Micro.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.```

BrentWilkins, 2016-04-23:
Please see issue #12.

stejsoftware, 2016-04-23:
Yea I figured out its actually a problem with the Arduino IDE. Once I found where the files were being installed i removed them all and tried again. It works now. You can close this ticket. Thanks!

closed 2016-04-23 by stejsoftware