sparkfun/Arduino_Boards #17: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude"

kallisti5, 2016-07-12:
```
Sketch uses 4,194 bytes (14%) of program storage space. Maximum is 28,672 bytes.
Global variables use 151 bytes (5%) of dynamic memory, leaving 2,409 bytes for local variables. Maximum is 2,560 bytes.
java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:115)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:160)
	at processing.app.debug.Compiler.upload(Compiler.java:163)
	at processing.app.Sketch.upload(Sketch.java:1220)
	at processing.app.Sketch.exportApplet(Sketch.java:1194)
	at processing.app.Sketch.exportApplet(Sketch.java:1166)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2487)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 11 more
```

Shit's broke yo.  Arduino 1.6.6, Fedora 24

Krastanov, 2016-07-26:
Confirmed on the hourly build from 25th of July.

BrentWilkins, 2016-08-02:
There are known bugs with Arduino. Does the new version of the IDE with its "lot of bugfixes to builder" fix the issue you are seeing? https://www.arduino.cc/en/Main/ReleaseNotes

Krastanov, 2016-08-02:
Seems to be working on my computer (Debian jessie, arduino 1.6.10)

kallisti5, 2016-08-02:
1.6.10 gives me a different error here:

"sorry - this program has been built without plugin support"

Non-sparkfun boards compile. Fedora 24, x86_64 + 1.6.10 downloaded from  Arduino site.

BrentWilkins, 2016-08-02:
@kallisti5 will you give this URL a try? https://raw.githubusercontent.com/BrentWilkins/Arduino_IDE_1.6.10_Testing/master/package_sparkfun_index.json

Your issue seems to be related to a new bug: https://github.com/arduino/Arduino/issues/5168

kallisti5, 2016-08-02:
That fixes it.  rm -rf ~/.arduino15; then ran through install with your new manifest.

BrentWilkins, 2016-08-02:
Thanks. It works on my system too. I'm going to replace the [official SparkFun version](https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json) with the version you just tested.

ToniCorinne, 2017-02-13:
@BrentWilkins can this be closed out?

BrentWilkins, 2017-02-13:
There have been 9 versions of the IDE released since the version used in the initial issue, and I pushed some changes back in August. Closing this issue.

closed 2017-02-13 by BrentWilkins

TinyChou, 2018-05-14:
I pull the latest version from master branch on my Mac OS and build the app with ``ant dist``
then run the .app just open the HelloWorld Example: Blink.ino: upload to my Uno board.

java.io.IOException: Cannot run program "{runtime.tools.avrdude.path}/bin/avrdude": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:207)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:732)
	at processing.app.SketchController.exportApplet(SketchController.java:703)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2197)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:248)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 10 more

[image]