sparkfun/Arduino_Boards #18: Incorrect nrfutil linking for Sparkfun nRF52832 Breakout

dcellucci, 2016-12-02:
Hello, 

I was having trouble compiling sketches using the Sparkfun board definition for the nRF52832, and I ended up identifying that the nrfutil executable linked in platform.txt was faulty. I am running OSX 10.10 with Xcode CLT 7.2 and Anaconda 4.1.6. I suspect it's because the version of nrfutil in this repository was built for a later version of OSX (it says it was built for 10.12)

I found that replacing the nrfutil in the Sparkfun/tools folder with a symlink to the executable whose path is found using `which nrfutil`.

ToniCorinne, 2017-02-13:
@jimblom think you could take a look at this?

dsanders11, 2017-07-28:
Ran into this same issue, 8 months later. It should be as simple as SparkFun simply rebuilding the package with the OS target being an earlier version like 10.9 or 10.10.