sparkfun/Arduino_Boards #23: SAMD21 mini: Error while burning bootloader

FRVisser, 2017-05-19:
```
Arduino: 1.8.2 (Mac OS X), Board: "SparkFun SAMD21 Mini Breakout"

{runtime.tools.openocd-0.9.0-arduino.path}/bin/openocd -d2 -s {runtime.tools.openocd-0.9.0-arduino.path}/share/openocd/scripts/ -f /Users/fokkovisser/Library/Arduino15/packages/SparkFun/hardware/samd/1.3.2/variants/SparkFun_SAMD_Mini/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; init; halt; at91samd bootloader 0; program {{/Users/fokkovisser/Library/Arduino15/packages/SparkFun/hardware/samd/1.3.2/bootloaders/zero/SparkFun_SAMD21_Dev.bin}} verify reset; shutdown 
java.io.IOException: Cannot run program "{runtime.tools.openocd-0.9.0-arduino.path}/bin/openocd": error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at java.lang.Runtime.exec(Runtime.java:620)
	at java.lang.Runtime.exec(Runtime.java:485)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:11)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:129)
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:412)
	at processing.app.Editor.lambda$handleBurnBootloader$14(Editor.java:2486)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 7 more
Error while burning bootloader.
```
The OpenOCD .path link does not work, `tools.openocd.path={runtime.tools.openocd-0.9.0-arduino.path}`. With the arduino Zero is is now `tools.openocd.path={runtime.tools.openocd-0.9.0-arduino6-static.path}`

Platform: OSX 10.11.6
Arduino IDE: 18.2
Sparkfun library: Sparkfun SAMD boards 1.3.2

Ah see now I have the wrong Arduino SAMD library: 1.6.15 (should be 1.6.9)

FRVisser, 2017-05-19:
https://github.com/sparkfun/Arduino_Boards/pull/24

jimblom, 2018-08-30:
Thanks for finding this @FRVisser! (Sorry for the delay.)

I just committed a fix for this along with a larger update to the samd core. Burn bootloader seems to be working on my end, feel free to re-open if it gives you any trouble.

closed 2018-08-30 by jimblom