sparkfun/Arduino_Boards #30: Broken Ternary Operation in lilypadusbplus pins_arduino.h

seanlaw, 2017-12-28:
I'm using the Lilypad Protosnap Plus setup and trying to load a super simple sketch:

```
#include <SoftwareSerial.h> //All I'm doing is including this header file

void setup() {
  //Setup nothing
}

void loop() {
  //Loop nothing
}
```
This results in the following error:

```
In file included from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/io.h:99:0,
                 from /Applications/Arduino.app/Contents/Java/hardware/tools/avr/avr/include/avr/interrupt.h:38,
                 from /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:41:
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp: In member function 'void SoftwareSerial::begin(long int)':
/Users/law978/Library/Arduino15/packages/SparkFun/hardware/avr/1.1.8/variants/lilypadusbplus/pins_arduino.h:159:219: error: expected ':' before ')' token
 #define digitalPinToPCMSKbit(p) ( ((p) == 1) ? 0 : ((p) == 21) ? 1 : ((p) == 22) ? 2 : ((p) == 23) ? 3 : ((p) == 6) ? 7 : (((p) == 4)||((p) == A4)) ? 4 : (((p) == 8)||((p) == A8)) ? 5 : (((p) == 12)||((p) == A10)) ? 6 )

                                                                                                                                                                                                                           ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:364:28: note: in expansion of macro 'digitalPinToPCMSKbit'
     _pcint_maskvalue = _BV(digitalPinToPCMSKbit(_receivePin));

                            ^
/Users/law978/Library/Arduino15/packages/SparkFun/hardware/avr/1.1.8/variants/lilypadusbplus/pins_arduino.h:159:219: error: expected primary-expression before ')' token
 #define digitalPinToPCMSKbit(p) ( ((p) == 1) ? 0 : ((p) == 21) ? 1 : ((p) == 22) ? 2 : ((p) == 23) ? 3 : ((p) == 6) ? 7 : (((p) == 4)||((p) == A4)) ? 4 : (((p) == 8)||((p) == A8)) ? 5 : (((p) == 12)||((p) == A10)) ? 6 )

                                                                                                                                                                                                                           ^
/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/libraries/SoftwareSerial/src/SoftwareSerial.cpp:364:28: note: in expansion of macro 'digitalPinToPCMSKbit'
     _pcint_maskvalue = _BV(digitalPinToPCMSKbit(_receivePin));

                            ^
exit status 1
Error compiling for board LilyPad USB Plus.
```

Basically, in the file pins_arduino.h (provided by Sparkfun) and line 159, the final nested ternary operator is missing a colon and an else operation. 

```
(((p) == 12)||((p) == A10)) ? 6 // missing something like "? 6:11" but I don't know what the else number should be
```

marshalltaylorSFE, 2018-01-11:
Thanks!  I missed that.  Update to version 1.1.9 and all should be good.

###Software serial compatible pins

* TX: 2,3,4,5,6,7,8,9,10,11
* RX: 4,6,8

closed 2018-01-11 by marshalltaylorSFE

seanlaw, 2018-01-12:
Can you please educate me on why TX can be on almost any of the pins but RX is limited to pins 4, 6, or 8?

marshalltaylorSFE, 2018-01-12:
Sure!  Software serial is accomplished by looking for changing edges on the input signal and recording the times, then converting to bits in a byte.  For the RX pin to work, it needs to A) have interrupt capabilities and B) be able to detect rising vs. falling edges.  Not all pins can do this.

Also, keep in mind that the software serial library is half-duplex, meaning that data can't be received while it is being sent out (also, the loop-back test doesn't work).

seanlaw, 2018-01-12:
Thanks for the info! I thought Pin 10 and Pin 11 had interrupts and therefore could be used for RX?

marshalltaylorSFE, 2018-01-13:
For the lilypad, pin 10 is assigned to port D, bit 0 and 11 is port D, bit 1, neither of which has the interrupt.  Maybe you're used to a different microcontroller?  For any micro in Arduino, the board pin name is mapped to some location on the processor, so between boards the underlying abilities of the pin can differ.

The pins_arduino.h defines the mapping.  The statement that was broken has to do with converting pin names to register locations on the processor with regards to the interrupts... which is why it broke when attempting to use software serial.

Hope this was revealing!