sparkfun/Arduino_Boards #31: add support for SparkFun Arduino Pro 328 (5V / 16Mhz) and (3,3V / 8Mhz)

yleny-dev, 2017-12-28:
Hello,

Can you add support for SparkFun Arduino Pro 328 (5V / 16Mhz) and (3,3V / 8Mhz) boards in the "Board definitions for SparkFun-manufactured AVR, ARM, and ESP-based Arduino boards. " software ?
* SparkFun Arduino Pro 328 (5V / 16Mhz)
https://www.sparkfun.com/products/10915
* SparkFun Arduino Pro 328 (3,3V / 8Mhz)
https://www.sparkfun.com/products/10914

I use Card Type : Arduino Duemilanove or Diecimila and Processor : ATmega328P in the Arduino IDE 1.8.5 for my SparkFun Arduino Pro 328 (5V / 16Mhz) board. It works with a basic example blink DEL.

I try to update the SparkFun boards.txt to add support for SparkFun Arduino Pro 328 with the following lines but it does NOT works:

################################################################################
################################### Arduino Pro 328 ############################
################################################################################
arduinopro328.name=SparkFun Arduino Pro 328

arduinopro328.upload.tool=avrdude
arduinopro328.upload.protocol=arduino

arduinopro328.bootloader.tool=avrdude
arduinopro328.bootloader.low_fuses=0xFF
arduinopro328.bootloader.unlock_bits=0x3F
arduinopro328.bootloader.lock_bits=0x0F

arduinopro328.build.board=AVR_ARDUINOPRO
arduinopro328.build.core=arduino:arduino
arduinopro328.build.variant=arduino:standard
arduinopro328.build.mcu=atmega328p

################################# Arduino Pro 328 3.3V ################################
arduinopro328.menu.cpu.atmega328p3V3=ATmega328P (3.3V / 8 MHz)

arduinopro328.menu.cpu.atmega328p3V3.upload.maximum_size=30720
arduinopro328.menu.cpu.atmega328p3V3.upload.maximum_data_size=2048
arduinopro328.menu.cpu.atmega328p3V3.upload.speed=57600

arduinopro328.menu.cpu.atmega328p3V3.bootloader.high_fuses=0xDA
arduinopro328.menu.cpu.atmega328p3V3.bootloader.extended_fuses=0xFD
arduinopro328.menu.cpu.atmega328p3V3.bootloader.file=atmega/ATmegaBOOT_168_atmega328.hex

arduinopro328.menu.cpu.atmega328p3V3.build.mcu.f_cpu=8000000L

################################# Arduino Pro 328 5V ################################
arduinopro328.menu.cpu.atmega328p5V=ATmega328P (5V / 16 MHz)

arduinopro328.menu.cpu.atmega328p5V.upload.maximum_size=30720
arduinopro328.menu.cpu.atmega328p5V.upload.maximum_data_size=2048
arduinopro328.menu.cpu.atmega328p5V.upload.speed=57600

arduinopro328.menu.cpu.atmega328p5V.bootloader.high_fuses=0xDA
arduinopro328.menu.cpu.atmega328p5V.bootloader.extended_fuses=0xFD
arduinopro328.menu.cpu.atmega328p5V.bootloader.file=atmega/ATmegaBOOT_168_atmega328.hex

arduinopro328.menu.cpu.atmega328p5V.build.mcu.f_cpu=16000000L

yleny-dev, 2017-12-28:
OK, I found the problem.
When I replace the lines 
arduinopro328.menu.cpu.atmega328p3V3.build.mcu.f_cpu=8000000L
arduinopro328.menu.cpu.atmega328p5V.build.mcu.f_cpu=16000000L
by
arduinopro328.menu.cpu.atmega328p3V3.build.f_cpu=8000000L
arduinopro328.menu.cpu.atmega328p5V.build.f_cpu=16000000L

It works.

mgrusin, 2017-12-28:
Hello, we're glad you solved your problem, but were you aware that the Pro and Pro Mini are already included in the base Arduino install? If you locate the "Arduino AVR Boards" section, "Arduino Pro or Pro Mini", with voltage / speed selectors, should be available to you.

jimblom, 2018-08-30:
As @mgrusin mentioned support for Arduino Pro's and Pro Mini's are included in the base Arduino AVR package.

[image]

closed 2018-08-30 by jimblom