sparkfun/Arduino_Boards #39: UART write routines are unbuffered.

WestfW, 2018-08-10:
The SAMD21 code is currently using a busy-loop for writing to UART devices. I consider that moderately unacceptable.
The Arduino core added txbuffer support in July 2017; it looks like the Sparkfun branch has not picked up that change. (You could do better than they did: I'm not thrilled that the Arduino core forces the use of the same-sized buffer for transmit and receive.)
arduino@b24c9e5#diff-9aa55b794319d1ec143466ac42b86e54

isaias-b, 2018-08-10:
I previously had the same issue with the mkrzero and the samd21 mini. However i didn't know that the compiled code of the sender was the bottleneck in my setup. I suspected it to be because on my mac it was possible to fetch data fast enough for my use case. Unfortunately the blocking caused a way larger performance penalty on the pi so that i was mislead and didn't even think about that the problem could lie in the senders code. After a long time of searching i could only see this assumption to be false and tada: https://forum.arduino.cc/index.php?topic=360286.msg2689347#msg2689347
Teensy compiles into non blocking calls on the usb.
I first ended up trying to solve this issue as described by @WestfW, but finally decided to just order new hardware which comes with the appropriate tools to compile more accurately to meet my requirements. This whole topic took me about 3w of my after work spare time to figure out. https://raspberrypi.stackexchange.com/questions/77081/data-transfer-of-uart-over-usb-too-slow

I will leave an answer on the Q&A pointing to this ;)

WestfW, 2018-08-10:
The lack of UART buffering, and the poor performance of bulk data over a USB link are separate problems...

isaias-b, 2018-08-20:
Well ok, then i am sorry for pointing it out 😶 😅

jimblom, 2018-08-29:
@WestfW Thanks for the heads up. I've updated the Arduino core in our samd variant to match what Arduino provides in their latest release (1.6.19). 

(This update picked up a lot of unrelated improvements as well. It was about time.)

The fixes should be available in v1.5.0 of our SAMD package.

(Message received re: static tx/rx buffer sizes. We'll see what we can do.)

closed 2018-08-29 by jimblom