sparkfun/Arduino_Boards #45: SparkFun SAMD21 Pro RF fails to compile with variant.h missing

alohawild, 2018-10-14:
Tried to uninstall and reinstall. Noticed now that the directory for the new board is not there. That is all the other boards have directories in "variants" and so guess we are missing that code.

alohawild, 2018-10-15:
Arduino: 1.8.7 (Mac OS X), Board: "SparkFun SAMD21 Pro RF"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/mwild/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/mwild/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/mwild/Documents/Arduino/libraries -fqbn=SparkFun:samd:samd21_proRF -vid-pid=0X1B4F_0X214F -ide-version=10807 -build-path /var/folders/jm/mpcd9zzw8xl7hl006s6fnn_80000gp/T/arduino_build_306581 -warnings=none -build-cache /var/folders/jm/mpcd9zzw8xl7hl006s6fnn_80000gp/T/arduino_cache_313943 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/mwild/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/mwild/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/mwild/Documents/Arduino/libraries -fqbn=SparkFun:samd:samd21_proRF -vid-pid=0X1B4F_0X214F -ide-version=10807 -build-path /var/folders/jm/mpcd9zzw8xl7hl006s6fnn_80000gp/T/arduino_build_306581 -warnings=none -build-cache /var/folders/jm/mpcd9zzw8xl7hl006s6fnn_80000gp/T/arduino_cache_313943 -prefs=build.warn_data_percentage=75 -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board 'samd21_proRF' from platform in folder: /Users/mwild/Library/Arduino15/packages/SparkFun/hardware/samd/1.5.1
Using core 'arduino' from platform in folder: /Users/mwild/Library/Arduino15/packages/SparkFun/hardware/samd/1.5.1
Detecting libraries used...
/Users/mwild/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10807 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x1B4F -DUSB_PID=0x214F -DUSBCON "-DUSB_MANUFACTURER=\"SparkFun\"" "-DUSB_PRODUCT=\"SFE SAMD21\"" -I/Users/mwild/Library/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/Users/mwild/Library/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/ -I/Users/mwild/Library/Arduino15/packages/SparkFun/hardware/samd/1.5.1/cores/arduino -I/Users/mwild/Library/Arduino15/packages/SparkFun/hardware/samd/1.5.1/variants/arduino_zero /var/folders/jm/mpcd9zzw8xl7hl006s6fnn_80000gp/T/arduino_build_306581/sketch/Blink.ino.cpp -o /dev/null
In file included from /Users/mwild/Library/Arduino15/packages/SparkFun/hardware/samd/1.5.1/cores/arduino/Arduino.h:81:0,
                 from /var/folders/jm/mpcd9zzw8xl7hl006s6fnn_80000gp/T/arduino_build_306581/sketch/Blink.ino.cpp:1:
/Users/mwild/Library/Arduino15/packages/SparkFun/hardware/samd/1.5.1/cores/arduino/delay.h:23:21: fatal error: variant.h: No such file or directory
 #include "variant.h"

                     ^
compilation terminated.
exit status 1
Error compiling for board SparkFun SAMD21 Pro RF.

jimblom, 2018-10-17:
Hi @alohawild .  Sorry about the trouble.

We've pushed an update to the SparkFun SAMD Boards -- 1.5.2  (with commit 4c6c02e). Updating your board definitions to that release should resolve this issue. Please re-open this issue if it does not.

closed 2018-10-17 by jimblom