sparkfun/Arduino_Boards #47: bossac fails on Windows 10

ShawnHymel, 2018-12-16:
Windows 10
Arduino 1.8.7
SparkFun SAMD Boards 1.5.2

Trying to upload to a SparkFun SAMD21 Mini Breakout, I get the following error:

```
Found upload port: COM15
C:\Users\sgmustadio\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.8.0-48-gb176eee/bossac.exe -i -d --port=COM15 -U true -i -e -w -v C:\Users\SGMUST~1\AppData\Local\Temp\arduino_build_882835/samd_timer_01.ino.bin -R 
bossac.exe: extra arguments found
Try 'bossac.exe -h' or 'bossac.exe --help' for more information
An error occurred while uploading the sketch
```

The only workaround I have found so far is to download bossac 1.7.0 and put it in

`C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\arduino\tools\bossac
`
Then, change line 134 of 
`C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\SparkFun\hardware\samd\1.5.2\platform.txt
`to 
`tools.bossac.path={runtime.tools.bossac-1.7.0.path}`

Not sure exactly what has changed in bossac 1.8.0, but that command does not seem to like it.

jimblom, 2018-12-20:
Hey Shawn!

This is a weird one...

Arduino's SAMD core still (in their SAMD core 1.6.20, at least) installs bossac 1.7.0 (see [http://downloads.arduino.cc/packages/package_index.json](http://downloads.arduino.cc/packages/package_index.json) and CTRL+F for "1.6.20"). It does looks like Adafruit's SAMD boards are packaged with a 1.8.0-48-gb176eee version of bossac, though, and, when installed, it goes into the "packages/arduino/tools" directory. Maybe that's confusing the IDE.

I can recreate this issue if I:

* Install the Adafruit SAMD board defs (1.2.9)
* Uninstall the Arduino SAMD board defs (1.6.20)
* Reinstall the Arduino SAMD board defs (1.6.20)
* Try to upload -- at this point it tries to use bossac v1.8.

Outside of modifying platforms.txt, I can fix this issue if I:

* Change board to Arduino/Genuino Zero (Native USB Port)
* Upload the sketch
* Change board back to SparkFun SAMD Dev
* Re-upload sketch

This somehow "corrects" the IDE to use the Arduino-packaged version of the tool instead of the Adafruit one. From here on no board changes seem necessary.

I'm curious why it's failing with the "extra arguments" error on 1.8, but the best fix for now may be specifying the version number of bossa, as you've found in your fix. Looks like Arduino does this in their SAMD platforms.txt. I'll work on getting something committed and pushing a new version to the board manager.

jimblom, 2018-12-20:
@ShawnHymel Give the 1.5.3 SparkFun SAMD board defs a try. Should hopefully fix the issue.

Please feel free to open this back up for any similar issue!

Thanks again, Shawn! Take care!

closed 2018-12-20 by jimblom

xsk8rat, 2018-12-22:
Fixed for me! Thank you!