sparkfun/Arduino_Boards #48: Unable to install the Blynk board

cmfrancis, 2019-04-29:
`"url": "http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz"` 
found in our JSON file returns a 404 and that breaks the ability to add the Blynk board to the IDE.

bboyho, 2019-05-01:
It looks like the server/domain name possibly changed on ESP8266's side? Older issues filed on the ESP8266 core for Arduino seems to indicate that that was the issue. User's will see this in the Arduino Boards Manager:

    Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz

Removing the download linked to to repo for the ESP8266 .json and it looks like something was updated about 22 days ago? Maybe we can update our files and update links based on this part of the .json file:

[https://github.com/esp8266/esp8266.github.io/blob/master/stable/package_esp8266com_index.json#L1080](https://github.com/esp8266/esp8266.github.io/blob/master/stable/package_esp8266com_index.json#L1080)

thirtytwobits, 2019-06-10:
Still seeing this issue. I can't develop against my blynk board right now.

djboby72, 2019-07-14:
There is a problem in the json file, so the file
win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
can not be downloaded.
You can change the json file by replacing the path
http://arduino.esp8266.com/
with
https://github.com/esp8266/Arduino/releases/download/2.3.0/
or you can use my modified file to add the Sparkfun BlynkBoard in arduino. But beware of the removal of generic support esp8266 if it is already installed because it is incompatible with Sparkfun BlynkBoard.
http://djboby.com/download/package_sparkfun_github_index.json

santaimpersonator, 2019-07-20:
You can only install either the generic esp8266 board definitions or SparkFun's esp8266 board definitions. Installing both board definitions cause a file path conflict since the SparkFun Thing Dev is named in both packages. (**Issue may include other ESP8266 board definitions.**)

I tested this on a few systems (Windows and Mac) and sometimes both folders are included, but other times the folders get overwritten by the new board definitions. Overall, the Arduino IDE is only able to see one of the board definition folders (or package) in either situation.

I created a [pdf with instructions](https://github.com/sparkfun/Arduino_Boards/files/3413135/Blynk-.Community.ESP8266.Boards.pdf) for a teacher, a few months back, to remove all definitions and only install SparkFun's board definitions to work around the issue. (*Sorry, I haven't had the time to update the tutorial. I hope this solves a majority of the issues; any feedback on whether this works is appreciated.*)

edspark, 2019-10-30:
We've updated our Arduino boards JSON file to fix the problematic link, update to 2.1.1 for the latest: 
[image].

bboyho, 2019-10-30:
@edspark updated the url and we were able to download the board definitions! Were were able to get **blink.ino** and **BlynkBoard_Core_Firmware** to upload with ESP8266.

edspark, 2019-10-30:
@santaimpersonator I've opened a seperate [issue](https://github.com/sparkfun/Arduino_Boards/issues/55) since this particular behavior is separate from the original issue.

closed 2019-10-30 by edspark

bboyho, 2019-11-14:
For those that are interested, @edspark worked with espressif to include the board in the [ESP8266 Arduino add-on [v2.6.0] ](https://github.com/esp8266/Arduino/releases/tag/2.6.0). We'll be updating the documentation as well.

🔥🔥🔥🔥🔥🔥🔥🔥🔥🔥🔥