sparkfun/Arduino_Boards #50: Wire.requestFrom() should not be followed by Wire.endTransmission() in Qduino.cpp

Koepel, 2019-05-17:
In the files:
- Arduino_Boards/sparkfun/avr/Qduino/Qduino.cpp
- Arduino_Boards/sparkfun/avr/libraries/Qduino/Qduino.cpp

in the function "readFrom()", the `Wire.requestFrom()` is followed by a `Wire.endTransmission()`. That `Wire.endTransmission()` may be removed. It is at the moment at line 284 (for both files).
Explanation: [Common-mistakes#2](https://github.com/Koepel/How-to-use-the-Arduino-Wire-library/wiki/Common-mistakes#2)

loricrotser, 2019-11-05:
Reviewed and committed. If you get a chance to double check, it'd be appreciated. Thank you!

closed 2019-11-05 by loricrotser

Koepel, 2019-11-05:
It is okay.

When the sensor is not connected, the readFrom() does not fill the lsb and msb parameters. Thus those values will keep their default value which is set to zero before calling readFrom(). That is a good way to deal with a sensor that might not be connected.