sparkfun/Arduino_Boards #51: bossac fails on Windows 10

CevG33ko, 2019-06-24:
VersionSpecs:
Windows 10
Arduino 1.8.9
SparkFun SAMD Boards 1.6.1
Arduino SAMD Boards 1.8.1

Trying to upload to a SparkFun 9DoF Razor IMU M0, I get the following error:

 `java.io.IOException: Cannot run program "{runtime.tools.bossac-1.7.0.path}/bossac.exe": CreateProcess error=2, `...

I found that the variable  `runtime.tools.bossac-1.7.0.path` is now called `runtime.tools.bossac-1.7.0-arduino3.path` in arduinos SAMD boards plattform.txt 

Two fixing alternatives found:
 -  Update Sparkfun SAMD by changing line 134 of
`C:\Users\%USERNAME%\AppData\Local\Arduino15\packages\SparkFun\hardware\samd\1.6.1\platform.txt` from : `tools.bossac.path={runtime.tools.bossac-1.7.0.path}` to `
tools.bossac.path={runtime.tools.bossac-1.7.0-arduino3.path}` and restart the arduino IDE to reload files.
- Downgrade the installed Arduino Board to version 1.6.19 on your system (this one worked for me and seems to be latest still supported version)

Please update your guide if you can reproduce this Issue since I followed the listed steps for the problem to arise (https://learn.sparkfun.com/tutorials/9dof-razor-imu-m0-hookup-guide/all#installing-the-9dof-razor-arduino-core)

pepelepew71, 2020-01-04:
my VersionSpecs:
Ubuntu 16.04 (x86 64 bits)
Arduino 1.8.10
Arduino SAMD Boards 1.8.4
SparkFun SAMD Boards 1.8.4

I got the same error message. It can be fixed by downgrading Arduino SAMD Board to version 1.6.19 , and SparkFun SAMD Boards to version 1.6.1 .

bubba198, 2020-12-25:
@pepelepew71 Thank you. Same issue here:

`Cannot run program "{runtime.tools.bossac-1.7.0-arduino3.path}/bossac.exe": CreateProcess error=2, The system cannot find the file specified` - same here, your fix to downgrade corrected the issue, what a mess is this Pro RF board