sparkfun/Arduino_Boards #57: Arduino 1.8.9/1.8.10, Openocd path needs update again

chipperdoodles, 2019-10-31:
Similar/Same to #23 

I had trouble burning the bootloader for any SFE samd21 boards and replaced the OpenOCD sketch upload section of sparkfun samd21 package platform.txt, with the the one from the aruduino samd platform.txt and that seemed to have allowed me to burn bootloaders.

tldr: 

* sparkfun package path: 
tools.openocd.path={runtime.tools.openocd-0.9.0-arduino6-static.path}

* Arduino package path:
tools.openocd.path={runtime.tools.openocd-0.10.0-arduino7.path}

edspark, 2019-12-04:
Sorry I missed this and thank for putting the perfect amount of information in this issue. I'll close this out when I get a fix in sometime this week, or early next.

edspark, 2020-06-03:
Fixed [here](https://github.com/sparkfun/Arduino_Boards/commit/3083af8031d4d02b0d95c8f27827fef3e625e4cb), thanks!

closed 2020-06-03 by edspark