sparkfun/Arduino_Boards #59: SparkFun Qwiic Micro SAMD21 - High Sleep Current

adamgarbo, 2019-11-06:
Hi there,

This morning I received the SparkFun Qwiic Micro - SAMD21 Development boards I ordered. The first thing I did was to measure the quiescent draw when in deep sleep and I noticed immediately that something was up. After cutting the power LED trace, I placed the board into deep sleep using the [Arduino Low Power library](https://github.com/arduino-libraries/ArduinoLowPower). I ended up measuring a sleep current draw of ~350 uA and an active current draw of ~9.0 mA. This definitely gave me a sense of déjà vu, as I encountered [the very same issue](https://github.com/adafruit/Adafruit_SleepyDog/issues/17) with my SAMD-based Adafruit Feather boards.

The good news it's a simple fix! The following lines of code need to be commented out in the wiring.c file in the SAMD board definition (lines 83-87). The SAMD21 pins are tri-stated by default, which is their lowest power consumption configuration (Section 42.6 of SAMD21 manual). Commenting out (or removing) these lines of code prevents the digital and analog pins from being set to INPUT mode.

```
// Setup all pins (digital and analog) in INPUT mode (default is nothing)
for (uint32_t ul = 0 ; ul < NUM_DIGITAL_PINS ; ul++ )
{
  pinMode( ul, INPUT ) ;
}
```
After doing so, I was able to achieve a sleep current of ~60 uA, which is about as low as you're going to get with this board and the AP2112 LDO.

Cheers,
Adam

edspark, 2019-11-13:
This is great info! I'd like to put this into our hookup guide - giving credit to you, if you don't mind?

adamgarbo, 2019-11-13:
Hi Ed,

I don't mind at all! Better yet, should this become a commit to the SparkFun board definitions? Adafruit ended up [making the same change](https://github.com/adafruit/ArduinoCore-samd/commit/bd082fa1f613cfad6b8cf0eb6eebce31005b2881) to their SAMD board definitions. Unless there's some reason I'm not aware of that requires the pins to be initialized as INPUTs?

Cheers,
Adam

edspark, 2019-11-13:
That is a good point, I don't see a reason for that to be in there.... I'll update after a conversation with some coworkers around here. Thanks again @adamgarbo :+1:.

edspark, 2019-12-04:
I can not believe it's been 28 days - I guess there was a holiday in there 🤔 . I've released a new build or our Arduino boards that has these changes implemented. Thanks again 👍 !

closed 2019-12-04 by edspark