sparkfun/Arduino_Boards #71: Micromod SAMD51

benevpi, 2020-11-05:
This may be intended behaviours (or I may be doing something a bit wrong), but it looks like there are some pin definitions missing from the Micromod SAMD51

Running the Example1_DisplayTest from the HyperDisplay 4DLCD-320240 Arduino Library with the samd51 in the Input And Display board, I get errors that D1, D0 and PWM0 aren't defined. It runs fine when the values are set as follows:

```
#define PWM_PIN PIN_A2             // Pin definitions
#define CS_PIN 0
#define DC_PIN 1
```

benevpi, 2020-11-06:
Just to add to this, I don't think the buzzer pin for the Input and Display micromod board (PWM1) is defined at all. I've been able to access the buzzer with:

`#define BUZZERPIN PIN_A2+1`

edspark, 2020-11-06:
First thanks for the issue! So in the Arduino IDE, your regular digital pins (D0, D1, D2) you call using the number *without* the letter D. 

For example:
```C++
pinMode(0, OUTPUT); 
digitalWrite(0, HIGH);
digitalWrite(0, LOW); 
```
The use of the `#define`  that you've mentioned in your first comment is a totally viable solution. You might also see something like
```C++
int csPin = 0; 
int dcPin = 1; 

pinMode(csPin, OUTPUT)
digitalWrite(csPin, HIGH);
digitalWrite(csPin, LOW); 

// Or however these pins are used, this is just an example
```
As to your second comment, you've found *a* solution but one that isn't exactly user friendly. What you've done is called the value `PIN_A2` that is set in the `variant.h` file and then incremented to `A3` by adding one which conincidentally is the next entry in the `PinDescription` list. That will not always work, though it will work most of the time. You can simply call `A3`:

```C++
pinMode(A3, OUTPUT);
```
`A3` is the pin number for PWM1 on the SAMD51. Here's a picture from the schematic:

[image]

Note that you're able to use "A3" to call an analog pin where you would **not** be able to call a digital pin in the same manner "D3". If you have anymore questions about pin numbering please check out our hookup guide on the SAMD51 MicroMod: 

https://learn.sparkfun.com/tutorials/micromod-samd51-processor-board-hookup-guide/hardware-overview

This post did make me realize that those PWM pins may be nice to call directly, and I'll get that changed so that in the future you can do a:

```C++
#define BUZZERPIN PWM1
```

closed 2020-11-06 by edspark