sparkfun/Arduino_Boards #79: Incomplete pin definitions for Sparkfun SAMD51 Thing PLus

ajahern95, 2021-05-09:
The pin definitions does not fully line up with the pinout of the SAMD51 Thing Plus pin out, as seen here: https://cdn.sparkfun.com/assets/c/a/8/6/8/ThingPlusSamD51a.pdf

For example, D9 the interrupt and analog is not attached, nor analog on D4. D9 is important on the Adafruit Music Maker Feather Shield, as it is used as an interrupt to allow background music playback.

Original line for D9 in variants.cpp:
{ PORTA,  7,  PIO_DIGITAL, PIN_ATTR_DIGITAL, No_ADC_Channel, NOT_ON_PWM, NOT_ON_TIMER, EXTERNAL_INT_NONE }

Possible corrections:
{ PORTA,  7,  PIO_DIGITAL, PIN_ATTR_DIGITAL, **ADC_Channel7**, NOT_ON_PWM, NOT_ON_TIMER, **EXTERNAL_INT_7** }.

santaimpersonator, 2023-05-10:
I wan't to say that we intentionally left out some of the additional pin functionalities for a specific reason(s), but I don't remember why. *(That was back in 2019. Maybe pin compatibility... like it could only be defined as a digital pin or analog pin?)*

However, I do remember we linked in the [Adding More SERCOM Ports for SAMD Boards tutorial](https://learn.sparkfun.com/tutorials/858) to the SAMD51 Thing Plus hookup guide. This was for specific use cases, where users needed to modify the board definitions. Maybe we could clarify that more in the hardware overview or Arduino IDE sections.

* It looks like we only have it linked at the end of the paragraph in the Hardware Overview section:
  [image]
* Maybe, if we linked the SERCOM tutorial like the board definition tutorial from the [ESP32 guide](https://learn.sparkfun.com/tutorials/esp32-s2-thing-plus-hookup-guide/software-overview):
  [image]

Thoughts?

ajahern95, 2023-05-11:
The issue I had, that made me look into pin definitions was I couldn't get the Adafruit music maker featherwing with amp (Adafruit 3436) to work.

It ended up being due to an interrupt that was not defined for D9 on the Sparkfun SamD51 thing plus. Without this, the music maker doesn't work, and was required within the Adafruit library, but it wasn't mentioned in any documentation.

This missing definition was the most important, the rest was me just comparing and making the Sparkfun SAMD 51 pin definition equal to the Adafruit one.

[image]<https://user-images.githubusercontent.com/36016723/237250130-43b5f872-2333-4711-bd16-c80d064b99ed.png>
[image]<https://user-images.githubusercontent.com/36016723/237250940-24eef45b-984d-45cb-970c-79d7aa5fdf66.png>

edspark, 2023-05-22:
Easy enough fix, thank you for the issue @ajahern95 and bringing this to me @santaimpersonator .

closed 2023-05-22 by edspark