sparkfun/Arduino_Boards #86: SparkFun STM32 Boards Not Found

ilush-a, 2021-06-17:
The board manager for SparkFunSTM32 is not showing up as described here: https://learn.sparkfun.com/tutorials/stm32-thing-plus-hookup-guide 

The other SparkFun boards show up but the STM32 does not. I am not sure why it randomly disappeared and it seems like a common issue. I used this board last week with no issues. 

These errors show up: 
Index error: could not find referenced tool name=xtensa-esp32-elf-gcc version=1.22.0-80-g6c4433a-5.2.0 packager=esp32
Index error: could not find referenced tool name=esptool_py version=2.6.1 packager=esp32
Index error: could not find referenced tool name=mkspiffs version=0.2.3 packager=esp32
Index error: could not find referenced tool name=xtensa-esp32-elf-gcc version=1.22.0-80-g6c4433a-5.2.0 packager=esp32
Index error: could not find referenced tool name=esptool_py version=2.6.1 packager=esp32
Index error: could not find referenced tool name=mkspiffs version=0.2.3 packager=esp32

Similar issues but for the ESP boards discussed here: https://forums.adafruit.com/viewtopic.php?f=25&t=177076

lucas-flora, 2021-06-25:
this! thank god its not only me. what happened here?? We are using this board to prototype a production device and I need to get it up and running. Anyone have luck with the stm32 branch of this repo?

awende, 2021-06-25:
I'm not able to replicate the index errors for the ESP32, but I have been able to get the STM32 boards to not show up. My guess is they both have the same root cause though.

What URL do you have for our board manager in the Additional Board Manager URLs list? Is it this one:

https://raw.githubusercontent.com/sparkfun/Arduino_Boards/master/IDE_Board_Manager/package_sparkfun_index.json

Recently GitHub has moved away from having default branches as "master" and changed the default to "main", which we recently switched over to. The new URL is:

https://raw.githubusercontent.com/sparkfun/Arduino_Boards/main/IDE_Board_Manager/package_sparkfun_index.json

The old URL still works, but it looks like it's a cached version of the JSON which has V1.0.0 of the ESP32, but not newest V1.0.1, and the STM32 boards do not appear to be there either.

lucas-flora, 2021-06-25:
@awende wow ok thanks! I think that may have worked, verifying now. I will say this should immediately be updated [here ](https://learn.sparkfun.com/tutorials/installing-arduino-ide#board-add-ons-with-arduino-board-manager)to avoid further confusion!

awende, 2021-06-25:
Yeah, we're working on updating the tutorials. Our newer hookup guides point to that tutorial which will make updating it easier. It might take a little bit of time to see how many of our older tutorials provide the JSON link directly.

nseidle, 2021-06-25:
Thanks @lucas-flora, I had missed that one. Fixed now. If you see any others, please let us know.

lucas-flora, 2021-06-25:
@awende @nseidle awesome thank you both for the quick responses and the excellent little board :]

ilush-a, 2021-06-25:
Thanks everyone! The new URL is working great

closed 2021-06-25 by ilush-a