sparkfun/Arduino_Boards #100: Product Documentation

LeeLeahy2, 2022-07-10:
Please add a link to this GitHub repository in the documentation links of the associated products.

This will make it easier to find how to enable Arduino support (enable development) for the various products.

santaimpersonator, 2023-05-09:
We appreciate you feedback! I'll pass this information along to the appropriate teams.

We'll have to discuss making this change permanent on the product page. However, I agree that it is useful. Which product(s) were you having trouble finding this information on? I'll note to follow up on it.

***FYI -** We do have a tutorial on [installing board definitions](https://learn.sparkfun.com/tutorials/installing-board-definitions-in-the-arduino-ide) that is normally linked in the latest tutorials/hookup guides (past few years). It should include all the `.json`s needed for most of our boards.*

---

**Note *(Internal Only)*:**

Referring to something like the [ESP32 Thing Plus USB-C](https://www.sparkfun.com/products/20168):
[image]

santaimpersonator, 2023-05-10:
Unfortunately, it was decided to not include the the `.json` link on the product page.

With that said, we will be keeping this [tutorial](https://learn.sparkfun.com/tutorials/1265) updated and the `.json` should be provided in the hookup guide of the associated product.

closed 2023-05-10 by santaimpersonator