sparkfun/Arduino_Boards #103: Sparkfun nrf52 library missing

Rubines, 2022-12-07:
Hi

I followed the tutorial fro the Sparkfun nRF52832 Breakout board, but the Sparkfun nrf52 library is not installed anymore

https://learn.sparkfun.com/tutorials/nrf52832-breakout-board-hookup-guide#adding-arduino-compatibility

SFE-Brudnerd, 2023-02-02:
Hi @Rubines, please ensure you're using the correct branch of the repo. I'm working to pull that branch in, but for now please use that.

santaimpersonator, 2023-05-10:
@SFE-Brudnerd Just to keep this tracked... once you have pulled the `nrf5` branch in, can you file an issue on the [Installing Board Definitions in the Arduino IDE tutorial](https://learn.sparkfun.com/tutorials/1265) to add the `nrf52` to the list? If you don't know how to do that, just tag me on this issue. Thanks!