sparkfun/Arduino_Boards #115: Update arm-none-eabi-gcc for SAMD boards

stonehippo, 2023-11-23:
The 1.8.11 version bundles a fairly old version of the arm-none-eabi-gcc (excerpt below).

```
            {
              "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "4.8.3-2014q1"
            },
```

This is the version that was packaged with older versions of the Arduino SAMD board support. However, newer versions have update this toolchain:

```
            {
              "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "7-2017q4"
            },
```

It would be great to have the Sparkfun Arduino BSP updated to more recent versions. Among other things, this would fix install issues on some platforms, including Raspberry PI/aarch64, where there is no package available for the older 2014 gcc.

edspark, 2023-11-30:
Thanks for bringing this to our attention. I'll get this updated sometime today.

edspark, 2023-11-30:
I updated the version used in the package json in the latest release 1.8.12, and I think the IDE is smart enough to use it since it's bundled with Arduino's SAMD packages and our release depends on theirs.  if I've missed something, please don't hesitate to let me know!

closed 2023-11-30 by edspark